openscopeproject/InteractiveHtmlBom #3: Silkscreen font

f6fvy, 2018-07-26:
Great tool ! Thanks for the job !

It looks like the font displayed on the silkscreen image is a serif-font, while the font used in pcbnew (and on the actual PCB) is a sans-serif one.

A very minor inconvenience, but I wanted to let you know.

Running on Win7 / Firefox

qu1ck, 2018-07-26:
That's weird. Can you post a screenshot? You can crop it to just show a piece of text.
Is it not monospace?
https://github.com/openscopeproject/InteractiveHtmlBom/blob/5ac0b1ed6685f24288abc22b81a70cf56ef5c23b/InteractiveHtmlBom/ibom.html#L510

Problem with pcbnew font is that it does not exist in ttf format, it's specially crafted for the purpose of being easily readable monospace font with low vector segment count.
I had to choose something that looks somewhat similar to it and has similar proportions. Monospace seemed like a good fit and had decent fallback on linux which doesn't have it.

f6fvy, 2018-07-26:
[image]

qu1ck, 2018-07-26:
That looks like Courier New. I agree it's not the best.
If you change the line I linked from Monospace to Consolas, does it look better?

I will try to find a way to have better fallback.

f6fvy, 2018-07-26:
Consolas is fine !

[image]

Thanks for your time !

commit 7a05c9a4 (2018-07-27): Add better fallback fonts in pcb rendering

Fixes #3
closed 2018-07-27 by qu1ck