openscopeproject/InteractiveHtmlBom #5: Standalone python on Linux fails with errors

jslater89, 2018-07-27:
Using Mint 19.

```
$ python2 ~/development/kicad/InteractiveHtmlBom/generate_interactive_bom.py libra-cervisiae-pcb.kicad_pcb
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxCommandEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxNotifyEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxScrollEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxScrollWinEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxMouseEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxKeyEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxSizeEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxPaintEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxNcPaintEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxEraseEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxMoveEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxFocusEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxChildFocusEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxCloseEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxShowEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxMaximizeEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxIconizeEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxMenuEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxJoystickEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxDropFilesEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxActivateEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxInitDialogEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxSetCursorEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxSysColourChangedEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxDisplayChangedEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxUpdateUIEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxNavigationKeyEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxPaletteChangedEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxQueryNewPaletteEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxWindowCreateEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxWindowDestroyEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxHelpEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxContextMenuEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxMouseCaptureChangedEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxMouseCaptureLostEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
[... cut ...]
No PointCount method on outline object. Unpatched kicad version?
No PointCount method on outline object. Unpatched kicad version?
No PointCount method on outline object. Unpatched kicad version?
No PointCount method on outline object. Unpatched kicad version?
No PointCount method on outline object. Unpatched kicad version?
No PointCount method on outline object. Unpatched kicad version?
No PointCount method on outline object. Unpatched kicad version?
Dumping pcb json data
start: Name "com.ubuntu.Upstart" does not exist

```

qu1ck, 2018-07-27:
I don't see anything here that would suggest it failed.
You can ignore the wx asserts.
`Dumping pcb json data` means bom was generated. Do you not have a bom folder next to pcb file?

As for the last error `start: Name "com.ubuntu.Upstart" does not exist` I believe I fixed this in master. Can you try latest git version? I haven't cut a new release yet.

qu1ck, 2018-07-28:
Note: bom folder is created where the pcb file is, not in working directory.

jslater89, 2018-07-29:
Latest on master branch works correctly.

Thanks! It's really an excellent tool.

closed 2018-07-29 by jslater89