openscopeproject/InteractiveHtmlBom #6: Arrow key behaviour

Gasman2014, 2018-07-27:
Expected behaviour - using arrow up or down should scroll through BOM list sequentially, item by item.

Actual behaviour (at least on macOS) is to move cursor in correct direction but by an indeterminate amount. I would consider this functionality to be handy for use when populating a board as it would not require the user to go from looking through a microscope to positioning the mouse cursor and back.

This is not critical but confusing! 

Otherwise very impressed (and impressed you fixed my previous bug v quickly too ;) Thanks for this excellent bit of software ).

Sembazuru, 2018-07-28:
Gasman2014's observed behavior seems to be OS dependent. On Win10 on both Firefox and Chrome the arrow keys don't do anything at all. Also the highlight in the BOM goes away when the mouse isn't over the BOM row (both browsers).

Gasman2014, 2018-08-11:
> Closed #6 <https://github.com/openscopeproject/InteractiveHtmlBom/issues/6> via 439806e <https://github.com/openscopeproject/InteractiveHtmlBom/commit/439806ebf8636059fe5fc0dd48eda595a746e602>.

Thank you - and FYI works fine in Safari too.