openscopeproject/InteractiveHtmlBom #7: Add controls to only show front/back components in bom

qu1ck, 2018-07-27:


Sembazuru, 2018-07-28:
With this, I would suggest changing the viewing control for the board representation. Have a 3-way button (like the current board view) to show F, FB, and B in the BOM, and a 2-way button for the board view for "active BOM view" and "both sides". 

Or, I suppose, for simplicity sake, use the same button set for both. When the board view is front, only show the front components in the BOM. When board view is back only show back components. And, when both sides on board view show all the components.

qu1ck, 2018-07-28:
Yeah, I was planning to use existing F/FB/B buttons to filter bom as well. Seems simple and effective.

commit c4b827e2 (2018-07-28): Clean gutters on load

Fixes #7
commit ad8969ce (2018-07-29): F/FB/B buttons switch bom tables too

Fixes #7
closed 2018-07-29 by qu1ck