openscopeproject/InteractiveHtmlBom #10: Look into improving html layout on mobile devices

qu1ck, 2018-07-28:
Rendering on narrow screens is a major PITA. You can't discern anything useful on pcb render on tiny screen but at least the top buttons should be visible so that you could switch to BOM only mode.

Sembazuru, 2018-07-28:
I haven't had too much problem seeing a board on my phone, but I've got a large screen (Pixel-xL), more like a phablet.

Here is a board in Android Chrome that I have with my phone in landscape, no other adjustment.
[image]

Here is the same board (also in Chrome) where I selected just the top of the board, changed the frame setup for portrait, and then in Chrome dragged the frame separator to trade white space around my board for space for the list.
[image]

Ok, I can agree that the reference designators on the board view can be hard to read, but there are plenty of other landmarks to find where the red square on the screen refers to on the actual board.

My big issue is it is easier to get access to the layout if it is all a single file. I'll open up another issue for that because it doesn't actually fit this one well.

qu1ck, 2018-07-28:
Yes, as it is viewing bom on a phone is not impossible, it just is not optimized at all.
For example on my moto g6 (1080x2160) screen is not wide enough to even show the layout buttons in portrait mode, you have to go landscape. And fonts are way bigger than necessary on high ppi screens, wasting space.