openscopeproject/InteractiveHtmlBom #12: Default settings when all components are on a single layer

f6fvy, 2018-07-28:
Suggestion : When all components are on one layer only (usual case of THT components, for example), display this layer by default on page opening (F or B button). In this case, we may even wonder if it is really useful to offer the choice to the user to display an empty layer ?

Sembazuru, 2018-07-28:
There is a minor advantage, but I can see it being a corner case. Because the solder points are marked on the THT pads it can be a guide to making sure one hits all the solder points. Especially useful for components that have short legs (ICs, sockets (chip carriers), and connectors come to mind easily).

You can turn it off it really bothers you or you need the extra screen real estate. It just isn't automatic.

qu1ck, 2018-07-28:
I will not remove ability to view an "empty" layer since you can still see highlighted through hole pads and at the very list it may help visualize component position when looking at the board from the soldering side.

Regarding defaults I am planning to add few command line options and also a dialog window, you will be able to set default view there. But this will be post v1.

qu1ck, 2018-08-02:
With dc04dec you now only have to change view once, it will be stored in your browser local storage.
Stored settings are tied to pcb title and revision, if you change them the view will be reset.

qu1ck, 2018-10-23:
This is implemented now with config dialog and command line options. No single commit to link here, it was a bunch of work since v1.1 tag.

closed 2018-10-23 by qu1ck