openscopeproject/InteractiveHtmlBom #14: BOM list sorting

Sembazuru, 2018-07-28:
What is the natural sorting of the BOM list? I can't seem to see any patterns.

My enhancement suggestion would be to allow clicking on the column headers to set the sorting and toggle sort direction.

qu1ck, 2018-07-28:
```python
    # sort table by reference prefix and quantity
    def sort_func(row):
        qty, _, _, rf = row
        ref_ord = {
            "R": 3, "C": 3, "L": 1, "D": 1,
            "J": -1, "P": -1
        }.get(rf[0][0], 0)
        return -ref_ord, -qty
```

It sorts by component prefix and within that by quantity. Every component type is divided in groups in order of decreasing priority:
Caps and resistors
Inductors, diodes
Others
Connectors, jumpers, test points
Within that components are sorted in decreasing order of quantity.

I never really paid attention to this, but looks like not mixing together caps and resistors, inductors and diodes would be better.

I will probably make default sorting configurable post v1.
Not sure about clicking column headers, I'll look into it.

hildogjr, 2018-07-31:
Connector "CNN" are being order by "C".
The order (of natural assembly) would be "R", "C", "L" (strict), "U", "Q", others..., "CNN", "J", "P".

qu1ck, 2018-08-03:
CNN order fixed in d72816ee

hildogjr, 2018-08-03:
If this new version:

Traceback (most recent call last):
  File "/usr/share/kicad/scripting/plugins/InteractiveHtmlBom/generate_interactive_bom.py", line 442, in Run
    main(pcbnew.GetBoard())
  File "/usr/share/kicad/scripting/plugins/InteractiveHtmlBom/generate_interactive_bom.py", line 412, in main
    pcbdata["bom"]["both"] = generate_bom(pcb)
  File "/usr/share/kicad/scripting/plugins/InteractiveHtmlBom/generate_interactive_bom.py", line 93, in generate_bom
    bom_table = sorted(bom_table, key=sort_func)
  File "/usr/share/kicad/scripting/plugins/InteractiveHtmlBom/generate_interactive_bom.py", line 75, in sort_func
    prefix = re.findall('^[A-Z]+', rf[0])[0]
IndexError: list index out of range

Starting pcbnew by terminal and using the plugin I got some
     No PointCount method on outline object. Unpatched kicad version?

hildogjr, 2018-08-03:
Will be nice to add to the order:
C, R, L, D, Q, U,
"SW" -> swithces
Y, X, F, S
"A" -> Some modules os KiCad
CNN
"HS" -> heat sink
"MH" -> to really the end, it is used as mounting hole
J, P
"NT" -> net tie, usually a component to join different nets and completely ignored on the assembly

qu1ck, 2018-08-03:
> prefix = re.findall('^[A-Z]+', rf[0])[0]
> IndexError: list index out of range

Fixed in d88f82e

> Unpatched kicad version

Read about custom shape pads in readme. If you install tomorrows nightly that will go away and you will see custom shape pads and copper/silkscreen graphics in html render.

hildogjr, 2018-08-03:
I got some error like 'not possible to load _pcbnew.kiface' (after replace it).
Maybe because I am using Linux + KiCad Nightly. Which use `pcbnew.py` named as 3.0.8.

Is this patch (named as 3.0.12) that you did to some workaround of some issue in `pcbnew.py`?

qu1ck, 2018-08-03:
> Is this patch (named as 3.0.12) that you did to some workaround of some issue in pcbnew.py?

Yes. But I only precompiled patched binaries for windows 64bit, you can't install them on linux.
Good news is my patch was accepted in KiCad mainstream so tomorrow's nightly will contain it and you can install one for your distribution.

qu1ck, 2018-10-23:
This is implemented now with config dialog and command line options. No single commit to link here, it was a bunch of work since v1.1 tag.

Arbitrary custom default component sorting is available with command line options and config dialog.

closed 2018-10-23 by qu1ck