openscopeproject/InteractiveHtmlBom #15: Preview BOM and unselect unwanted references.

Sembazuru, 2018-07-29:
Since you brought up plans for an interface, here is more clutter for it ;-)

Be able to see a preview of the BOM list to allow unselecting components. This would allow not even showing "do not install" parts, mounting holes, and miscellaneous logos.

This is just for a cleaner looking final output, shove this idea way down on the priority list.

qu1ck, 2018-07-29:
Hint: you can go to footprint properties and set placement type to virtual, in that case bom plugin will not count that component, although will render it's pads and silkscreen. Essentially that's your do not install part. 
Mounting holes are by default virtual, bom already doesn't count them.

qu1ck, 2018-10-23:
This is implemented now with config dialog and command line options. No single commit to link here, it was a bunch of work since v1.1 tag.

There is no preview but you can always generate bom first time to use as preview, configure component blacklist and generate it again.

closed 2018-10-23 by qu1ck