openscopeproject/InteractiveHtmlBom #16: Implement part number and description or custom field columns

qu1ck, 2018-07-29:
This info is not available in pcbnew. I will remove these fields until I think of a good way to pull it from eeschema.

qu1ck, 2018-07-29:
I gotta admit, I'm impressed by that typo in commit message. That's a quality typo.

hildogjr, 2018-07-31:
I think net file file haven't this information.
Maybe the BOM xml file from Eeschema, but is quite complicated (huge update on the plugin).
If you are insterested the file
https://github.com/hildogjr/KiCost/blob/master/kicost/eda_tools/kicad/kicad.py
provide the ability to read the XML BOM.

qu1ck, 2018-08-01:
Thanks for the pointer, I do prefer xml bom over eeschema file, since the format of the latter is subject to change soon.

drid, 2018-08-01:
Actualy it is contained in the net list.
From a new project i added a single R and added a "Part" field with value "GPT0603BLABLABLABLA".
The exported netlist contains:
```
  (components
    (comp (ref R1)
      (value R)
      (datasheet ~)
      (fields
        (field (name Part) GPT0603BLABLABLABLA))
      (libsource (lib Device) (part R) (description Resistor))
      (sheetpath (names /) (tstamps /))
      (tstamp 5B622043)))
```

hildogjr, 2018-08-03:
Depend of the the NET list file may be a bad business, now Pcbnew doesn't need it to synchronize with Eeschema.

qu1ck, 2018-10-23:
This is implemented now with config dialog and command line options. No single commit to link here, it was a bunch of work since v1.1 tag.

closed 2018-10-23 by qu1ck