openscopeproject/InteractiveHtmlBom #17: Checkbox for placed components

drid, 2018-07-29:
Add a checkbox on each row that will mark this component group as "placed'.

Makes it easier to keep track of component placement

qu1ck, 2018-07-30:
Wanted to do that too, but maybe more generic. Have "sourced" and "placed" checkboxes by default and make it configurable so that you can add/replace them with say "ordered" and "soldered".

drid, 2018-08-01:
Even better.
If possible add sortcuts like 'o' and 's' that will mark the component and move to the next line

qu1ck, 2018-08-05:
This works now but I'm not entirely happy about how checkboxes are edited.
There is not enough validation and it would be nice to be able to rename a checkbox without loosing the ticked references data.

hildogjr, 2018-08-07:
When marked as placed could have some transparency to the parts or same way to identify on the 2D model that the components were placed.

qu1ck, 2018-08-11:
I can't think of a good way to mark components on pcb. What happens if you mark same component with multiple checkboxes? "Placed" is not special in any way, it can be anything and it can be multiple things.
I'll leave it as is for now, unless someone has good ideas for UX.

TechJA, 2018-08-11:
@qu1ck .
just an idea:
clicking on component reference to mark it as placed.
[image]

qu1ck, 2018-08-11:
There are 2 problems with that:
1. component boundary boxes can overlap
[image]

You wouldn't be able to tell one component under another

2. Again, what about other checkboxes? Should only "placed" be marked, why? How to differentiate/mix the colors if multiple are checked?

And to take a step back, how is this helpful in the first place? Why would you want placed components be marked on the render, when you presumably have a real board with components you can clearly see placed.

I can see some value in the strike through text for the reference. If you have a long list of components in a group and you go and place them one by one, in order to not loose track of where you are in the list, marking them in some way would help. But frankly I would be more off-put by necessity to click the mouse every time I make a move with tweezers so that doesn't look like a good solution either.

TechJA, 2018-08-11:
It was just an idea to mark placed components, not for you to implement it.
Imagine this scenario: pcb with several items and several components for each item with packages 0402 or 0603. It will be better if you could mark each placed component. The components boundary boxes won't overlapped because each item  are only displayed when you select the line on the table. For example, you could click in one reference text and it change to strike through and the color of boundary box change from red to other color.

TechJA, 2018-08-12:
> Ah I see, so it would be only highlighted in different color when the row is highlighted. That makes sense

Exactly.

qu1ck, 2019-12-26:
BOM now can show ungrouped view so that you can tick checkboxes for each component individually. This is more generic solution to the problem discussed above. I think this can be closed now.

closed 2019-12-26 by qu1ck