openscopeproject/InteractiveHtmlBom #18: Dark mode

qu1ck, 2018-07-30:
Full blown theming support seems overkill but simple dark mode would be nice.

hildogjr, 2018-07-31:
This have to wait the GUI as you propose. In the GUI a comboBox could allow choose between dark / light theme, collapse references or not.

hildogjr, 2018-07-31:
I like to your way to keep just one file (because this I did #19). So maybe create different CSS styles on the plugin folder and merge it inside the html on the final process.

hildogjr, 2018-07-31:
Even the java script and other could be a external file added by the python code in the final HTML. This will allow some users to easy modify the HTML template adding some personal logo (example).
But most important you could keep the *.js file format indented and remove the '\n' in the moment before add at the html.

qu1ck, 2018-08-01:
I was thinking to have dark/light mode be a toggle in html itself so that you could change it dynamically.

On splitting js and css, that is a good idea, although not because of indentation. Js that is not indented is minified anyway.

qu1ck, 2018-08-01:
Rendering of the canvas is the biggest CPU/GPU consumer, parsing the file is nothing in comparison so I won't spend too much time on it. 
Also in the large board you posted in the other issue, 99% of the html file volume is the pcbdata, which is already compact as it is. Some corners can be cut there like using shorter string keys and reducing floating point precision, but ultimately not much, unless I switch to a compact serialization protocol, something like protobuf. But that is a huge dependency to pull in, which I will only do if wins are really justified.
Anyway, we digress :) Lets keep this bug about dark mode.

closed 2018-08-08 by qu1ck

hildogjr, 2018-08-08:
When in dark mode, the board edge should go white? Just a question about the better contrast to visualize.

commit 29fdeb10 (2018-08-08): Add dark mode

Fixes #18