openscopeproject/InteractiveHtmlBom #21: Not include not populate components

hildogjr, 2018-07-31:
Components if the DNP / NOPOP defined could be not listed on BoM.
Need to analyse how to get this information, on [KiCost tool](https://github.com/hildogjr/KiCost/blob/master/kicost/eda_tools/eda_tools.py) we get this information (of the fields defined) by the BOM xml file, but this could also be read directly from the sch file.

Sembazuru, 2018-07-31:
This is very similar to my whishlist item https://github.com/openscopeproject/InteractiveHtmlBom/issues/15

Also mentioned is a way to currently have DNP components skipped (I tested and this works) is to have the component in the board file defined as a "Virtual Component". This script already ignores virtual components. (This also allows you to have a 3D model already defined in the footprint but set it to virtual and you can filter those out in the 3D view of KiCad.)

qu1ck, 2018-08-26:
This will be implemented as part of #15

closed 2018-08-26 by qu1ck