openscopeproject/InteractiveHtmlBom #22: Zoom ability

TechJA, 2018-08-01:
First of all congrats for your excellent work!

Are you thinking of implementing the ability to zoom in/out on the PCB panels? Being limited to screen size is very limiting for those of us with small screens. Adjusting the panel size just isn't enough for larger boards.

qu1ck, 2018-08-01:
I will keep this as a wishlist, but I'm not sure how useful it is.
The main feature of this plugin is to quickly give an overview of all places on the board where a component is located. From that you can look at the board and either quickly find reference designator on silkscreen since you know where to look at, or compare with relative positions of other pads/silkscreen marks to quickly locate your footprint.
If you have board zoomed in and part of it is outside of the view, how would you know where to look?

Screen size is an issue but unless you are trying to do A4 size boards with 0603 components on 1080p monitor, it's not a big issue. In #20 a big board was posted, it can be viewed fine on 15" laptop.

hildogjr, 2018-08-06:
Nice work with the zoom @qu1ck. Just a comment about user experience, nothing to worry but the scroll is the opposite of the behavior in Pcbnew, is this the default behavior on java script?
About your changes in `pcbnew.py` I am waiting the Nightly release of KiCad, the compilation for Ubuntu is broken since 2 weeks ago (I am in contact with the support). So I will test in that big board and give you some feedback.

qu1ck, 2018-08-06:
What do you mean opposite? For me scrolling up zooms in both in pcbnew and in bom.
There is no "default" behavior, it's all hand coded.

hildogjr, 2018-08-06:
The zoom in / zoom out works in one direction when I roll the scroll button in the mouse in Pcbnew and in the opposite when roll in the iBOM generated HTML.
Even is the opposite direction of the default browser zoom (rolling to front, that is the default OS zoom in makes zoom out in your page).

Using: Firefox v61 on Ubuntu.
Is not really a bug, but I release that was zooming in when I wanted zoom out, and vice-versa.

qu1ck, 2018-08-06:
Yeah, for me both chrome and firefox zoom same way.
Can you tell me does it zoom in or out when you scroll up? Does direction change if you use chrome?
I suspect it's pcbnew doing something in non-standard way and has reverse direction, not browser.

qu1ck, 2018-08-06:
Also do you have "natural scrolling" or something like that enabled in linux? I.e. when you scroll up does the page move up or down?

TechJA, 2018-08-06:
qu1ck,
They're right (Firefox on Windows). Scrolling up zooms out, scrolling down zooms in. It should be the other way around.

qu1ck, 2018-08-06:
Reopening to investigate further.

qu1ck, 2018-08-06:
Yeah I have math totally wrong on zoom calculation. Will fix soon.

hildogjr, 2018-08-06:
No problem. Our function is help debugging.

qu1ck, 2018-08-07:
I believe this should be fixed now. Please try the latest.

TechJA, 2018-08-07:
@qu1ck ,
it's possible lock the color of the select line when you move the cursor to PCB's panels?

[image]

Could you implement a feature  (ex: add a button) to  recenter the the pcbs and reset the zoom to full view?

qu1ck, 2018-08-07:
> Could you implement a feature (ex: add a button) to recenter the the pcbs and reset the zoom to full view?

Right click does that.

> it's possible lock the color of the select line when you move the cursor to PCB's panels?

Yes, that's doable.

TechJA, 2018-08-07:
> Could you implement a feature (ex: add a button) to recenter the the pcbs and reset the zoom to full view?

> Right click does that.

confirmed - Thanks!

hildogjr, 2018-08-07:
Yes @qu1ck, so much smoothly the zoom now! (both: mouse and touch pad)
Now mouse wheel works as the conventional direction of my OS Ubuntu (zoom on Firefox, Chrome, Pcbnew and others software having the same direction).
The touch pad is working in both too (before wasn't), but in the opposite direction (in both), the "Natural scrolling touckpad configuration" have no effect on this.

hildogjr, 2018-08-08:
I tested on other computer (Linux Firefox also) but with the zoom on the lateral of the trackpad and worked fine (direction and "quantity").
The other computer (first one) use the 2-fingers zoom and the behavior may be related with some really specific hardware-software combination. May be not interesting use so much time in this correction, so.

qu1ck, 2018-08-08:
So I tried this on a macbook pro with touchpad. Woah it's smooth! And inverted. But looking at the events browser is firing there is really nothing I can do about it, it looks just like mouse wheel rotated in opposite direction.
I suspect 2 fingers zoom on linux would be the same but I can't confirm. Laptops that have separate scrolling area probably work same as actual mouse wheel, that's why it works for you.

I'm going to call this one fixed, if anyone has ideas on how to improve detection of touchpads, feel free to drop a line here or send a pull request.

closed 2018-08-08 by qu1ck

qu1ck, 2018-08-11:
@TechJA your last ask to lock highlighted row was implemented.

TechJA, 2018-08-11:
@qu1ck ,
thanks!

commit 9f41fa73 (2018-08-05): Add continuous redraw on drag setting

Fixes #22