openscopeproject/InteractiveHtmlBom #23: Alternate print mode with all component group renderings on separate pages

qu1ck, 2018-08-01:
Current print mode is limited in only printing long list of BOM or BOM + current rendering on a single page.

It's useful to have a way to quickly print out all component group renderings.

mattimo, 2020-06-21:
Having this feature would enable a great workflow using an eReader like the reMarkable to assemble boards more or less distraction free.

What'd be the best way to implement this? Would it be just a "print" media css or should the generator just generate a PDF directly?

qu1ck, 2020-06-21:
My current thinking is that integrating this functionality into existing html output is not a good fit.

In the future I plan to add support for custom html templates where different variety of pages optimized for different use cases can be generated that have same underlying functionality of pcb rendering, bom grouping, etc, but have different UI and purpose. Html page with a render for each component group can be one such template.

Also to make this happen rendering code needs to be refactored into something reusable and not as tightly coupled to html as it is right now.