openscopeproject/InteractiveHtmlBom #27: Copies of reference/ value field are not handled

poeschlr, 2018-08-07:
In KiCad one can add a copy of the value and reference fields with user text fields that have the text %R, REF**, %V or VAL**. This tool does not handle this at all. (%R is still displayed as %R instead of being displayed as the reference assigned to the footprint in question)

This feature is heavily used in the official library. (Every footprint has a copy of the reference field on the fab layer)

qu1ck, 2018-08-07:
Interesting, I didn't really notice this. Will fix.
Are there other substitutions or is that all of them? Is a comprehensive list documented somewhere?

qu1ck, 2018-08-08:
Hm I can't find a way to repro this. When components on pcb are actually linked to schematic component they seem to automatically pick up the right text and are displayed correctly in bom render.

Only time when I get REF** on bom is when pcbnew also shows REF**, i.e. I just plopped a footprint directly in pcbnew instead of importing it from schematic so it does not have a reference or a value.

Can you add some steps on how to get incorrect text in bom, which is different from what is in pcbnew?

poeschlr, 2018-08-08:
I have the main reference of my personal footprints on the fab layer and the copy on the silk layer.

Here the screenshots of my small test project:
[image]
[image]

And the test project it self.
[iactive_bom_test.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/2270095/iactive_bom_test.zip)

My kicad version is:
Application: kicad
Version: (5.0.0), release build
Libraries:
    wxWidgets 3.0.3
    libcurl/7.54.1 OpenSSL/1.0.2l zlib/1.2.11 libssh2/1.8.0 nghttp2/1.23.1 librtmp/2.3
Platform: Windows 7 (build 7601, Service Pack 1), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.3 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.60.0
    OpenCASCADE Community Edition: 6.8.0
    Curl: 7.54.1
    Compiler: GCC 7.1.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_ACTION_MENU=OFF
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

I am running windows 7

qu1ck, 2018-08-09:
Thanks for great bug report!

commit bff64b39 (2018-08-09): Handle substitutions in footprint text

Fixes #27
closed 2018-08-09 by qu1ck