openscopeproject/InteractiveHtmlBom #28: Problem with alignment targets

jp-ricaud, 2018-08-08:
Hi,

It seems that InteractiveHtmlBom is not able to process the file if alignment targets are placed on the Edge.Cuts layer:

```
c:\Program Files\KiCad\lib\python2.7\site-packages\wx-3.0-msw/wx/_core.py:16633:
 UserWarning: wxPython/wxWidgets release number mismatch
  warnings.warn("wxPython/wxWidgets release number mismatch")
Loading c:\Program Files\KiCad\share\kicad\demos\sonde xilinx\test.kicad_pcb
Unsupported drawing class  PCB_TARGET
Unsupported drawing class  PCB_TARGET
Unsupported drawing class  PCB_TARGET
Dumping pcb json data
```

If the targets are removed, it works fine.

[image]

qu1ck, 2018-08-08:
It should just skip unsupported things. Are you sure bom was not generayed? Also you either didnt copy full log or you are using old version, please update to latest.

qu1ck, 2018-08-08:
Also it could be that you are trying to generate bom in program files which needs elevated priveleges. Try copying board file to normal directory.

qu1ck, 2018-08-09:
Bug confirmed, fix coming shortly.

commit a11a0573 (2018-08-09): Gracefully handle unsupported features

Also improve printed text to indicate that it's not an error and
unsupported parts are skipped.

Fixes #28
closed 2018-08-09 by qu1ck