openscopeproject/InteractiveHtmlBom #31: error on windows7 and python2.7

ogahozy, 2018-09-03:
i try installing this plugin to kicad 5.0 on 32bit windows 7 and i got this when i want to generate the BOM for my circuit:
Command error. Return code 2
Error messages:
C:\Program Files\KiCad\lib\python2.7\site-packages\wx-3.0-msw/wx/_core.py:16633: UserWarning: wxPython/wxWidgets release number mismatch
  warnings.warn("wxPython/wxWidgets release number mismatch")
usage: generate_interactive_bom.py [-h] [--nobrowser] file
generate_interactive_bom.py: error: unrecognized arguments: C:/Users/Harvey Bell/Desktop/nrf24-breakout/nrf24-breakout.
Expecting your replies

qu1ck, 2018-09-03:
Because you have a space in your path to the file you should enclose it in double quotes like this:

`....\python.exe "C:/Users/Harvey Bell/Desktop/nrf24-breakout/nrf24-breakout.kicad_pcb"`

Also unless you want to do bom generation from a batch script there is no reason to do it from console on windows. Just press the button in the pcbnew UI.

ogahozy, 2018-09-04:
this is what i have at the Ui command line:
python "C:\Program Files\KiCad\share\kicad\scripting\plugins\InteractiveHtmlBom/generate_interactive_bom.py" "%I" "%O".
this is the image
[image]

qu1ck, 2018-09-04:
You can't run this plugin like eeschema bom scripts. This is pcbnew action
plugin. Please read the instructions about installation and usage.