openscopeproject/InteractiveHtmlBom #33: Ref** not being placed

alpharesearch, 2018-09-04:
I have a PCB with a few holes that were placed in PCBNEW that just have the default Ref** designation. Looks like only the first of these different things shows up for all of them. I guess somewhere is a place to name these in PCBNEW, however I wanted to give feedback just in case you wanted to know.

alpharesearch, 2018-09-04:
Btw, I like your program. Thanks for developing it. Here is the file I used:

https://github.com/alpharesearch/ESP8266_DS18B20/blob/master/esp8266/esp8266.kicad_pcb

I'm using the default 5.0 Kicad Linux Mint version from the official web page. With the plugins link in the plugins directory PCBNEW crashed. But your program worked good from the command line.

qu1ck, 2018-09-04:
References are assumed to be unique so it's not unexpected that only one of REF** shows up. 
You should enumerate those footprints.

> With the plugins link in the plugins directory PCBNEW crashed.

This is more bothering, did you use the latest version? If you did and it still crashes, can you create the link, launch pcbnew from terminal (just type in pcbnew) and paste the output if any here?

qu1ck, 2018-09-04:
If you get a crash with plugin but no crash without plugin it definitely has to do with the plugin.

Can you paste output of these commands please:
`wx-config --version`
`wx-config --query-toolkit`

and this python script:
```python
import wx
print wx.version()
```

alpharesearch, 2018-09-05:
Here is the output from the commands:
$ wx-config --version
3.0.4
$ wx-config --query-toolkit
gtk2
python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import wx
>>> print wx.version()
3.0.2.0 gtk3 (classic)
>>> exit()
$ python3
Python 3.6.5 (default, Apr  1 2018, 05:46:30) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wx
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'wx'
>>> print wx.version()
  File "<stdin>", line 1
    print wx.version()
           ^
SyntaxError: invalid syntax
>>> exit()

qu1ck, 2018-09-05:
Right, so you have the same issue ubuntu 18 has, where maintainers had the brilliant idea to have wxwidgets built with gtk2 and wxpython built with gtk3. This causes python scripting in KiCad to be completely broken. Is it disabled in your KiCad build? If not that would explain the crash.
To check if it's enabled see the option for KICAD_SCRIPTING_WXPYTHON in kicad version info (about dialog -> show version info).

Fix for this that I found to work on debian testing is to downgrade wxpython to a version that is built against gtk2 (one that is in stretch).

alpharesearch, 2018-09-05:
After reading the readme file here I had this suspicion.
Makes sense because Mint 19 is based on Ubuntu. 
Thank for confirming.

alpharesearch, 2018-09-05:
However,  I personally still believe if Kicad has ambiguous references it would be good to cope in your program and still show the graphics correctly. Kicad it self has no issues, i would guess there is a real not user entered unique reference for each object. I would not rely on user entered references in this case. However,  for 99 percent this should not be an issue.

qu1ck, 2018-09-05:
I agree. At the very least the tool should render those components, even if they are not in the bom table. Maybe listing them in bom can be implemented too if I don't treat refs as unique. But obviously some features will not work properly like storing checkbox state per reference.
I'll keep this open, but it will not be a top priority.

closed 2018-11-18 by qu1ck

alpharesearch, 2018-11-18:
I just update to the latest version and now everything looks great. Thanks for fulfilling my wish.

commit b2b305f6 (2018-11-18): Switch to identifying modules by index instead of reference

Fixes #33, #57