openscopeproject/InteractiveHtmlBom #34: Way to donate? Patreon?

CRImier, 2018-09-05:
Hi! This is an awesome tool, works great for ZeroPhone PCB self-assembly =) How can I donate, or maybe support your initiative - PayPal? Patreon? Something else?

qu1ck, 2018-09-05:
Hi,
I'm glad you like this tool and I appreciate the thought, thank you.
But there is no need to support me, I do this completely for free.
If you wish to donate some money I can recommend donating to KiCad development
https://givetokicad.web.cern.ch/
I myself am not affiliated with Cern and I'm not a (core) KiCad dev, although I contribute sometimes.

CRImier, 2018-09-05:
Okay, let me know if you get a Patreon or something =) I'm talking about specifically this tool - I'll donate to KiCad as well, but this tool is a quality one, and solves the need our project has - keep us updated if a funding source appears =)

closed 2018-09-05 by CRImier