openscopeproject/InteractiveHtmlBom #37: Installation instructions inaccurate for macOS

therealprof, 2018-09-05:
It took me quite a while to get it to work because the installation instructions seem quite inaccurate.

The mentioned plugin paths do not match:

> /Applications/kicad/Kicad/Contents/SharedSupport/scripting/plugins

This one is missing an ".app" somewhere. The correct path for the homebrew installed version would be:
`/Applications/Kicad/kicad.app/Contents/SharedSupport/scripting/plugins/`

> ~/Library/Application Support/kicad/scripting/plugins

According to the Scripting Console, this is not in the Python search path. In fact the only path outside of the application itself is:
`~/Library/Python/2.7/lib/python/site-packages`

Linking it to the first folder above did not work for me. It needed to be copied there so I could use it.

> Save the file and press the iBOM button on the top toolbar.

The iBOM icon only showed up after imported it once manually from the Scripting Console.

> If the button is not on the toolbar Tools -> External Plugins... -> Generate Interactive HTML BOM also works.

The `External Plugins` menu does not exist under `Tools`.

> python2 path/to/InteractiveHtmlBom/generate_interactive_bom.py path/to/board.kicad_pcb

This will not work for most users because the KiCad modules will usually not be in the search path of the System Python interpreter:
```
Traceback (most recent call last):
  File "InteractiveHtmlBom/generate_interactive_bom.py", line 3, in <module>
    import pcbnew
ImportError: No module named pcbnew
```

Using the linked system Python framework inside the `kicad.app` also doesn't seem to work.

I'm on `Version: (5.0.0-3-g5ebb6b6), release build` BTW.

Short story how I made it work:
* git clone https://github.com/openscopeproject/InteractiveHtmlBom.git
* cd InteractiveHtmlBom
* cp -r InteractiveHtmlBom/ /Applications/Kicad/kicad.app/Contents/SharedSupport/scripting/plugins/
* open pcbnew
   * Tools->Scripting Console
     * `from InteractiveHtmlBom import plugin`

qu1ck, 2018-09-05:
Thanks for the report. I'm wondering if some of your troubles are caused by homebrew installation instead of straight up downloading an official .dmg file and copying application from there.

Here is what I did for completely fresh install from official package.

    mkdir -p ~/Library/Preferences/kicad/scripting/plugins
    ln -s InteractiveHtmlBom ~/Library/Preferences/kicad/scripting/plugins/InteractiveHtmlBom

After this you need to reload kicad and that's it, plugin button is there in pcbnew, no tinkering in scripting console needed.

I couldn't make script work from command line though, wx would just refuse to load some dynamic library.
Looks like in newer MacOS some changes were made, including application support folder change to ~/Library/Preferences
But for the most part instructions work. I'll fix the discrepancies.

> The External Plugins menu does not exist under Tools.

That's because in release build action menu option is disabled.

therealprof, 2018-09-05:
> Thanks for the report. I'm wondering if some of your troubles are caused by homebrew installation instead of straight up downloading an official .dmg file and copying application from there.

The recipe pretty much does the same and it's still the official version, so it won't affect the compiled in paths, etc. 

> After this you need to reload kicad and that's it, plugin button is there in pcbnew, no tinkering in scripting console needed.

That's what I attempted at first but as you can see from the Python Console, `~/Library/Preferences` or `~/Library/Application Preferences` are not in the search path:

```
>>> import sys
>>> sys.path
['/Applications/Kicad/kicad.app/Contents/Frameworks/python/site-packages/wx-3.0-osx_cocoa', '/Applications/Kicad/kicad.app/Contents/Frameworks/python/site-packages/wx-3.0-osx_cocoa', '/Applications/Kicad/kicad.app/Contents/MacOS', '/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7', '/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin', '/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', '/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-scriptpackages', '/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', '/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', '/Applications/Kicad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages', '/Applications/Kicad/kicad.app/Contents/SharedSupport/scripting', '/Applications/Kicad/kicad.app/Contents/Frameworks/python/site-packages', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/python27.zip', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-scriptpackages', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', '/vagrant/build/python-dest/Library/Frameworks/Python.framework/Versions/2.7/lib/lib-dynload', '/Users/egger/Library/Python/2.7/lib/python/site-packages', '.', '/Applications/Kicad/kicad.app/Contents/SharedSupport/scripting', '/Applications/Kicad/kicad.app/Contents/SharedSupport/scripting/plugins', '/Applications/Kicad/kicad.app/Contents/SharedSupport/scripting/plugins/InteractiveHtmlBom']
```

> That's because in release build action menu option is disabled.

Well, that would be a great thing to mention then, no? ;)

qu1ck, 2018-09-05:
It doesn't have to be in sys path, it has to be in pcbnew plugin search path. You can verify what directories pcbnew is looking at by running this from scripting console:
```python
import pcbnew
print pcbnew.PLUGIN_DIRECTORIES_SEARCH
```

> Well, that would be a great thing to mention then, no? ;)

That is only true for MacOS release build and some linux distributions. And it's changed recently to be default on. I can't keep track of all builds, but I'll add some wording to indicate that that menu might not be accessible.

commit dd49cd3f (2018-09-07): Update installation instructions

Fixes #37
closed 2018-09-07 by qu1ck