openscopeproject/InteractiveHtmlBom #47: Kicad has segfault with Ubuntu 18.04 package install of python dependencies and kicad.

frohro, 2018-09-12:
This seems to be due to the wxpython problems of Ubuntu 18.04.  If I use pip install wxpython, it works, except that you cannot quit the kicad project manager.
Thanks for a nice tool!
Rob

qu1ck, 2018-09-12:
This was brought up a few times so I'll keep this bug open as `won't fix`.

The problem is that most new distribution package maintainers provide both wxwidgets built with gtk2 and gtk3 and apps can use either one (at least until gtk2 is completely phased out) but they only provide wxpython built for gtk3. When app using wxwidgets gtk2 tries to load wxpython it loads wxwidgets gtk3 libraries as well which don’t work together and result is SIGSEGV: https://bugs.launchpad.net/kicad/+bug/1762432

KiCad devs are working on fixing this. In the meantime use this plugin from command line or downgrade wxpython package version to one built against gtk2.
On debian testing I had to install python-wxgtk3.0 (3.0.2.0+dfsg-4) from stretch.

michalfapso, 2019-01-19:
I wasn't able to run InteractiveHtmlBom as a KiCad plugin, because of this issue, but it is perfectly working as a standalone script:
https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Usage#standalone-script 

It just took me some time till I found out it is usable as a standalone script, so I am posting it here :)

frohro, 2019-04-10:
I'm happy to report that the InteractiveHtmlBom generator works just great with version 5.1 of kicad on Ubuntu 18.10.  You need to remove the .pyc files if you ran it with python previously (at least I did).  This is a great add on for kicad!

qu1ck, 2019-04-10:
Yup, the expected move of KiCad to python3 and gtk3 is what solved the issue. I updated the title of the bug to reflect current state more accurately.

qu1ck, 2022-06-12:
No longer relevant.

closed 2022-06-12 by qu1ck