openscopeproject/InteractiveHtmlBom #49: ArchiveProject doesn't work in combination with InteractiveHtmlBom.

automesslar, 2018-09-17:
The plugin [ArchiveProject](https://github.com/MitjaNemec/Kicad_action_plugins/tree/master/archive_project) doesn't work when I have the InteractiveHtmlBom plugin in my scripting folder.
ArchiveProject works fine with other plugins so it seems like the problem is on the side of InteractiveHtmlBom.

qu1ck, 2018-09-17:
Documentation of that plugin states that it doesn't work with post 5.0 nightlies. What version are you running?
I highly doubt InteractiveHtmlBom affects in any way other plugins.

automesslar, 2018-09-17:
I tested it with a nightly from July 16 which was before 5.0 release and with the 5.0 release build from [http://downloads.kicad-pcb.org/windows/testing/](http://downloads.kicad-pcb.org/windows/testing/).
It seems really weird that there is some conflict between plugins, but when I remove the InteractiveHtmlBom folder from my scripting folder ArchiveProject works fine. When it is there Archive Project doesn't work, so I don't have another explanation.

automesslar, 2018-09-17:
I just did some more testing with different versions of InteractiveHtmlBom. It seems like version 0.7 and older don't have the issue, but version 0.8 and newer have it.

qu1ck, 2018-09-18:
Thanks for the report. Narrowing it down to version 0.8 helped find the culprit faster. Apparently python doesn't like when two different modules try to use default logger.

commit 401bc283 (2018-09-18): Fix logging clashing with other plugin's logging

Fixes #49
closed 2018-09-18 by qu1ck