openscopeproject/InteractiveHtmlBom #50: Option to hide 'placed' rows

SchrodingersGat, 2018-09-28:
Once an entire group of components has been placed (e.g. all 0.1uF caps) it would be nice to have an option to hide that row. This would really help with efficiency of work by removing visual clutter from the list.

qu1ck, 2018-09-28:
I will think if there is a nice way to add this in terms of UI but in the meantime you can sort by checkbox so that all "placed" components are at the end of the list for example.

obbardc, 2019-01-01:
It would be awesome to paint the components a different colour after they are marked as placed.

rjordans, 2019-01-02:
Came here with the same idea as @obbardc 

I think it would be great to be able to see in the overview picture which components are already on the board.  Especially on partially populated boards I expect this to be very useful.

xfdr0805, 2020-04-29:
> 在将组件标记为放置后，将它们涂上不同的颜色将是很棒的。

Good Idea!

qu1ck, 2020-04-29:
This was implemented in 374f71efd6c24471039e43cb027a15b2584bf08e
You see components greyed out in bom.

closed 2020-04-29 by qu1ck