openscopeproject/InteractiveHtmlBom #54: Support "do not populate" and variants

nickoe, 2018-10-07:
The Interactive HTML BOM  plugin is very nice. But I have noticed some features missing, so I kindly ask for this wishlist item.

I would like to see the Interactive HTML BOM tool to be able to hide entries marked as "do not populate" in the schematic.

I am using KiCost and here one can mark it with DNP something. See the doc https://kicost.readthedocs.io/en/latest/usage.html#do-not-populate-parts

On top of this it could be useful if it would support the variant scheme used by KiCost as well. But this might be too much for this issue. See https://kicost.readthedocs.io/en/latest/usage.html#schematic-variants

I have not investigated other methods used to mark this from other BOM tools.

I am currently on 401bc2831ac717a061de255e2ca18d76f7365263.

qu1ck, 2018-10-07:
Both of these are already on the wish list, see #15 and #51.

KiCost and KiBom treat variant a bit differently and (of course) have different fields. I will add something that should work for most cases for users of both KiCost and KiBom but it won't be identical implementation.

closed 2018-10-07 by qu1ck

nickoe, 2018-10-07:
Ahh, ok. I did try to look through the issues, but I didn't catch their terminology. Good to see you intend to support it in the future. Thanks.