openscopeproject/InteractiveHtmlBom #56: Not showing some parts

sennin32, 2018-11-11:
Platform and KiCad version used Win10 & kicad 5.0.1
What are the steps to reproduce the issue : simply generate
What is the observed behavior : some parts not in the list
What is expected behavior : all the part except virtual one should shows up

Thank you for this great tool.
It's the best bom generator I've ever tried on Kicad for me.

There's a problem that some of the part it not shows up.
I'm not sure why.
I use SMD as the type, just the same as other parts.
if you need more info to spot the issue.
pleas tell me what I should post.

sennin32, 2018-11-11:
hi,
after some testing.
I found it is because of layers.
I guess it is kind of conflict.
There's a part that put on front, but some of it's pad is on back layer.
I changed the footprint, and it will work.
It is work with build in bom generator without problem :)

qu1ck, 2018-11-11:
Sorry, I don't follow.
Did you make it work? What was the issue? Was it just that you needed to switch to FB mode so that footprints from other side also showed up?

qu1ck, 2018-11-18:
Closing for inactivity

closed 2018-11-18 by qu1ck

sennin32, 2018-11-20:
hi,
it is work now.
Cause the part is meant to be on the bottom layer.
So I create it with sip pad on top layer, but smd pads on bottom layer.
but that make it disappear in the 3d view.
I changed smd and dip parts on the same layer, it works.