openscopeproject/InteractiveHtmlBom #57: Panelized boards only highlights the first boards parts

Protoneer, 2018-11-15:
I have a panel with 4 copies of the same board.

The Interactive HTML Bom list picks up all the components. Ex: A single cap is listed 4 times in the list.

Unfortunately on the board display section only the first boards components are highlighted.

Otherwise this is an amazing plugin!!! Do you have a donation page?

qu1ck, 2018-11-15:
This is likely same issue as #33, that is they share underlying cause: only one of the components with same reference will be rendered.
Seeing as panelizing is more common use case I will prioritize this.

qu1ck, 2018-11-15:
Re: donations see #34 TL;DR is: thank you but donate to KiCad/CERN instead if you wish.

closed 2018-11-18 by qu1ck