openscopeproject/InteractiveHtmlBom #58: Failed to create a temporary file name

Gasman2014, 2018-11-21:
Great project and works very smoothly. I only haven problem, If I adjust the default settings I am unable to save them. If I press the 'save current settings' - I am getting the following kicad error

`can't open user configuration file.`
```
23:31:13: Failed to create a temporary file name (error 30: Read-only file system)
23:31:13: can't open user configuration file.
23:31:13: Failed to create a temporary file name (error 30: Read-only file system)
23:31:13: can't open user configuration file.
```
I am on macOS 10.12, automatically open in browser is disabled. 

I am unclear as to where the current settings are saved. I have tried making the interactivebom directory world writable but to no effect.

qu1ck, 2018-11-22:
Settings are saved next to plugin code (where generate_interactive_bom.py is). That directory should be writable for user you run the plugin with.
On MacOS do not put the plugin under `/Applications`, put it in user directory instead `~/Library/...`

qu1ck, 2018-11-22:
Just tested on MacOS Mojave, saving settings works fine. But my setup looks like this, I have the repo checked into ~/src/InteractiveHtmlBom and created a link:
`~/Library/Preferences/kicad/scripting/plugins/InteractiveHtmlBom` -> `~/src/InteractiveHtmlBom/InteractiveHtmlBom`. No special permissions manipulation required.

Gasman2014, 2018-11-22:
That is pretty much how I had it set up. I initially checked out the repo into a directory in ~ and put an alias to it to Applications/Kicad/contents/..etc. I had some problems with this and ended up having to move the actual directory. I then tried the same with the path that you suggested. Neither location seemed to be able to save the preferences - both reporting the same error. i removed links and actual files from both locations, restarted and finally put the repo back in ~/Library/Preferences/kicad/scripting/plugins/. It now seems to work. There does seem to be a delay in how Kicad registers the plugins. Aliases don’t seem to behave quite the same as symlinks it would seem.

Anyhow, now seem to be working OK. I will experiment further with using a proper symlink.

 Thank you for your help.

closed 2018-11-22 by Gasman2014

qu1ck, 2018-11-22:
Yes, aliases are like shortcuts in windows. They operate on application level (Finder) not on OS/filesystem level like symbolic and hard links. Which means for most programs, including KiCad, an alias is just a file instead of a pointer to another directory.