openscopeproject/InteractiveHtmlBom #59: [Question] Does it have to run from the inside of KiCAD

zen, 2018-11-21:
I'm trying to use this amazing plugin in my automated CI/CD environment. Currently it's only possible to launch it from KiCAD. Would it be possible to make it fully standalone?

qu1ck, 2018-11-22:
Did you read the usage doc? It describes exactly how to do what you are asking.

https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Usage

closed 2018-11-22 by qu1ck

zen, 2018-11-25:
Of course I did. On fresh docker container it tries to import wx (which is not CLI only tool should do)
`python2 path/to/InteractiveHtmlBom/generate_interactive_bom.py path/to/board.kicad_pcb
python2 InteractiveHtmlBom/generate_interactive_bom.py /work/drone-kicad-sample.kicad_pcb
Traceback (most recent call last):
  File "InteractiveHtmlBom/generate_interactive_bom.py", line 10, in <module>
    import wx
ImportError: No module named wx
`

After installing Total 250MiB of extra packages I'm getting about 300 lines of that error:
`./src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxCommandEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?`

And the at the end:

`Unable to access the X Display, is $DISPLAY set properly?`

Obviously it's looking for X server which is not something I would expect from CLI command. As you can guess, CI/CD environments are usually headless servers, so no X server running there.

I think it might be just a simple thing to be removed, but I'm not python nor wx expert to try to do it. Any suggestions?

zen, 2018-11-25:
Ok, got one step further. I installed fake X server and was able to use the script. But still, it requires graphical environment to be running, which is probably easy to remove form the code.