openscopeproject/InteractiveHtmlBom #62: Remove parts with empty (or ~) value

omogenot, 2018-12-08:
Hi,
First of all, thanks for this tremendous job. 
I see that you added some kind of sorting/filtering items which is helpful when a board can have several versions (which is my case where a board has 10 different options).

However, it might be elegant to remove form the list (or add this as an option) all components which value field is empty (in fact Kicad does not allow empty fields and uses a tilde ~ character instead). To be able to un-populate the board BOM for a specific option.
Best Regards,
OM.

qu1ck, 2018-12-09:
I think what you are trying to achieve is already doable using board variant field.
Just add a field called "variant" (or anything else you like) and select that field as board variant field in extra fields. Then just tick all the variants you want in whitelist OR tick all the variants you don't want in blacklist, thats it.
If you want R1 to be both in variant1 and variant2 of the board, then set R1's variant value to (for example) "v1 v2" and make sure to tick that value as well as just "v1" in the whitelist when you generate bom for v1. Then for v2 you tick that "v2" and "v1 v2" so R1 is still included.

omogenot, 2018-12-15:
Fine by me.
Did I mentioned that the white and black lists are empty when I select a version variant? At least on my OSX 5.01 version of Kicad.

qu1ck, 2018-12-15:
Did you regenerate xml/netlist file after you populated version variant field?
If file is up to date try to reproduce the issue with a trivial board with just a few components, does it still happen?

omogenot, 2018-12-15:
Of course.

Please find hereunder two screenshots. One on OSX and one on Windows XP of the very same file. As you can see, the white and black lists are empty on OSX.

I do agree with you that everyone has a personal way of making variants. However, it would be nice to be able to hide value column as this could be misleading as a variant can eventually be based on using different component values...
Anyway, do not spend too much time on this.

By the way, would it not be wise to rename the resulting html file after the chosen variant, allowing to generate multiple iBom.html files at once.

[image]
[image]

qu1ck, 2018-12-16:
Thanks, I pushed a fix for variant field on mac.
Regarding hiding value field maybe at some point I will make all default columns optional/configurable.