openscopeproject/InteractiveHtmlBom #67: output file should be named after board file

Sembazuru, 2019-01-17:
Currently the output html file is always called ibom. If one has a project with multiple boards and wants to collect all the ibom html files into one location for documentation convenience then one has to manually rename the html files.

Also, if one uses (abuses?) KiCad to have subsheet boards like how I describe on the forum [here](https://forum.kicad.info/t/components-off-pcb/14707/14), I would need to rename the ibom html file to the board name each time I use this script, otherwise only one board layout will be preserved.

qu1ck, 2019-01-17:
I was planning to solve this using substitutions like %f - board filename %b - board project name %c - company %d - date etc. So you would configure output to be say "bom/%b.html".

Sembazuru, 2019-01-18:
Good to know it is the works, and a more flexible implementation than what I was thinking. Thanx.

qu1ck, 2019-01-21:
@Sembazuru wanna test the latest?

[image]

Sembazuru, 2019-01-21:
I just checked it out and it looks good. I didn't do much to try to break it. I was at first disappointed that the setting didn't seem to be persistent, until I remembered about the save current settings in the bottom of the window...

FWIW, I tested this in the Feb20 Win64 Nightly.