openscopeproject/InteractiveHtmlBom #72: Html page flexibility: header or footer, links?

rusefi, 2019-02-03:
Is there any way to control or customize (header or footer?) html output in any way? For instance I would like IBOM html to have a link to the relevant wiki or forum, is that possible?

qu1ck, 2019-02-03:
Top 4 fields (title, company, revision, date) are generated from pcb metadata. In Pcbnew go to `File -> Page Settings...` and you can modify those fields. Is that enough?

rusefi, 2019-02-03:
This is a lot, but neither of them is a URL so if my user clicks on the BOM there is no way for him to navigate from it. I have a feeling that a lot of people are not aware of the "Back" button.

Would custom static html header be easy to add or not really?

qu1ck, 2019-02-03:
You can put html in any of the fields and it works, example for an url:

`<a href="http://myforumurl.com">My forum</a>`

Adding custom static header is doable in theory but it's a whole can of worms of it's own. Making arbitrary headers fit into the page without breaking layout is hard.

rusefi, 2019-02-03:
I am 99.9% OK with that solution and you are right that custom static header is a can of worms.

Thank you for your amazing tool and prompt responses.

qu1ck, 2019-02-03:
No problem.
May I add your board to list of demos? This one
https://github.com/rusefi/rusefi/tree/master/hardware/frankenso

rusefi, 2019-02-03:
Absolutely! Just added a few more of our smaller boards and created a nightly refresh https://rusefi.com/docs/ibom/

commit c71cebc6 (2020-05-01): Add custom js css and header/footer support

Fixes #125
Fixes #72
closed 2020-05-01 by qu1ck