openscopeproject/InteractiveHtmlBom #73: Error running in python3

diggit, 2019-02-19:
Running **InteractiveHtmlBom** from cli using python3, I got error:
```
AttributeError: 'str' object has no attribute 'decode'
```

Solution was easy, just remove `.decode('utf-8')` from 2 lines. I know that such change breaks python2 compatibility. Not sure about proper solution.

qu1ck, 2019-02-20:
Thanks, I have a fix ready, will test tomorrow on gtk3/wxphoenix machine.

qu1ck, 2019-02-20:
Fixed in e594337

closed 2019-02-20 by qu1ck

Shackmeister, 2019-02-20:
@qu1ck Download newest version, now I get this

[image]

qu1ck, 2019-02-20:
That was not related to python3 but that is now fixed too.

Shackmeister, 2019-02-20:
my bad :) thank you!

qu1ck, 2019-02-20:
No worries, thanks for reporting bugs.

diggit, 2019-02-21:
Awesome, thanks for fixing.