openscopeproject/InteractiveHtmlBom #74: Feature request: Add option to indicate or color "placed" components on preview

nickajeglin, 2019-02-22:
It would be cool if there was an option to do that. Then the preview would look more like the real live board as you populate.

qu1ck, 2019-02-22:
Something like this is already planned, but I didn't intend to make visual indicator for placed components to be permanent, only when row is highlighted.
See discussion in #17.

I think permanent indication will only add clutter. Checkboxes already show state for the whole row. When you want to drill down on particular component group you highlight the row and see which particular ones are placed.

nickajeglin, 2019-02-22:
I didn't see the previous discussion in #17. That makes sense. I'd say we can close this one in that case.

closed 2019-02-22 by nickajeglin