openscopeproject/InteractiveHtmlBom #76: Option to render bounding boxes instead of pads and silkscreen

stiff, 2019-02-25:
On boards tightly populated with small-sized components silkscreen sometimes is quite ambiguous, and with pads everywhere it's hard to see the component names. It's more convenient to use bounding boxes and references, like it's usually done in *.Fab layers.

qu1ck, 2019-02-25:
Note: you can turn off silkscreen in ibom if you don't use it.

I'll keep this issue open as feature request to add fab layer option, this was requested multiple times.

closed 2019-04-08 by qu1ck

stiff, 2019-04-08:
Tried it and it works awesome, thanks!

The only slight inconvenience with it is that we can't easily toggle values like in Plot dialog, but probably it's asking too much :)

qu1ck, 2019-04-08:
I will probably add a "Hide all text" toggle to help with decluttering. It will hide silkscreen text too, does that work?

stiff, 2019-04-08:
Not sure what's the point of hiding all text, and if I'd saw it in UI it would be very unclear to me what it does. But maybe I just misinterpreting your idea, so here is my use case. For some projects I want to have separate BOM that maps references to part numbers and assembly plan like I attached in first comment. While for simpler PCB and prototypes, it's desirable to have just values and where to put them. So approach used in Plot dialog with 2 checkboxes "plot values" and "plot references" seems very logical and would work ok here.

qu1ck, 2019-04-08:
My idea was to basically drop all text rendering, that includes references and values and other text on all layers.
But I see your point, having separate knobs for showing values and references seems useful. I'll file a feature request for that.

commit 833369f4 (2019-04-08): Add option to hide pads

Fixes #76