openscopeproject/InteractiveHtmlBom #80: Open directory button "..." does not open the last folder

hildogjr, 2019-03-11:
On Ubuntu 16.04 KiCad Nightly
Using the open directory button "..." to specify the bom destination, it does not open the last folder at the the textbox. Instead open in a generic location.

qu1ck, 2019-03-11:
Works correctly for me on Debian with GTK3 and I remember testing it on GTK2 as well.
Did you press "save current setting" after changing directory? Otherwise on every plugin run default directory will be changed to whatever is in your settings.

hildogjr, 2019-03-11:
Yes, after change and save works as expected.
But I want to say in the first use, the GUI starts with the project (Pcbnew file folder) at the textbox and I was expecting that the dialog showed up by "..." start from it (it just happen after a first change and save).

hildogjr, 2019-03-11:
Curiosity: Where (each file) the saved settings are kept?

qu1ck, 2019-03-11:
Ah yes, default destination folder is ./bom/ (relative to pcbnew file location) and if it doesn't exist GTK folder selection widget goes to $HOME or whatever it thinks of while on windows and mac it does sensible thing - goes to first parent of the directory that exists, which will be pcbnew file directory.
I don't think this warrants a fix since there is an easy workaround already.

Config is kept in config.ini file in the root of the plugin directory. It's global, not per project.

hildogjr, 2019-03-11:
Ok. So was just a misunderstood mine, no issue here.

closed 2019-03-11 by hildogjr