openscopeproject/InteractiveHtmlBom #82: Add option to show datasheet field as extra field

qu1ck, 2019-03-13:
Datasheet field is a built in and needs special handling because it's stored differently.

hildogjr, 2019-03-13:
May use as hyperlink in a existent column at generated iBOM.

closed 2019-06-02 by qu1ck

eeintech, 2022-08-23:
@qu1ck Would you consider adding the "Datasheet" field in the list of fields since the format update in KiCad V6 doesn't differentiate it from other properties?

```
(property "Value" "CAP-000008-00" (id 1) (at 0 -15.24 0)
        (effects (font (size 1.27 1.27)) hide)
      )
      (property "Footprint" "Capacitors:C0402" (id 2) (at 0 -22.86 0)
        (effects (font (size 1.27 1.27)) hide)
      )
      (property "Datasheet" "http://xxx.com/" (id 3) (at 0 -13.97 0)
        (effects (font (size 1.27 1.27)) hide)
      )
```

qu1ck, 2022-08-24:
It is stored similarly to other properties in the schematic but ibom doesn't read schematic file. It reads either netlist or pcb file. Datasheet field is not exported to the pcb so you have to generate netlist file if you want to use datasheet field.

eeintech, 2022-08-24:
@qu1ck Yes it does work exporting the netlist. Follow-up question: can certain field be treated as hyperlink inside the HTML table?

In summary I would like this:
[image]

To become:
[image]

I could do a simple regex on the HTML, but I wonder if this feature is built-in somewhere?

eeintech, 2022-08-24:
Actually never mind on the Regex... I didn't realize the data was compressed.

qu1ck, 2022-08-24:
You can do any kind of postprocessing you want using [customization](https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Customization).

eeintech, 2022-08-25:
Thanks, I'll take a look!

eeintech, 2022-08-26:
Alright figured it out, even with my poor JS skills :sweat_smile: 

For anyone looking to add datasheet links to the BOM table, using KiCad V6:
1. Go to schematic, click on "File > Export > Netlist..." and export netlist
2. Go to layout, in the menu bar click on "Preferences > Preferences > Action Plugins" and find out what is the folder used for the InteractiveHtmlBom plugin. Open that folder.
3. Copy `user.js` file inside folder "web/user-file-examples" to the "web" folder, along with other `*.js` files
4. Edit this file and add the following javascript code:
``` js

function urlify(text) {
    var urlRegex = /(https?:\/\/(www\.)?[-a-zA-Z0-9@:%._\+~#=]{1,256}\.[a-zA-Z0-9()]{1,6}\b([-a-zA-Z0-9()@:%_\+.~#?&//=]*))/g;
    return text.replace(urlRegex, function(url) {
        return '<a href="' + url + '" target="_blank">' + url + '</a>';
    })
}

window.addEventListener('load', function() {
    var table = document.getElementById("bomtable");
    
    for (var i = 0; i < table.rows.length; i++) {
        table.rows[i].innerHTML = urlify(table.rows[i].innerHTML)
    }
});
```
6. Go back to KiCad layout, open InteractiveHTMLBOM plugin, select "Fields" tab, click "Show" checkmark next to `Datasheet` field (`Datasheet` field should be present in the list, else you've missed step 1).
7. Click "Generate BOM"
8. Enjoy hyperlinks!

@qu1ck Would you mind adding a link to this comment to this section of the doc, if you find it useful?https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Customization#examples

qu1ck, 2022-08-26:
Your code will work only once when page is loaded. Any time bom table is changed (with filtering or display mode or checkbox change) your links will disappear because the table is regenerated.

You should add a listener to BOM_BODY_CHANGE_EVENT event, the see user.js example.

eeintech, 2022-08-29:
Thanks, I have update my code snapshot above to make it a callback instead :+1:

qu1ck, 2022-08-29:
Added to wiki.

itsboo07, 2023-01-11:
> For anyone looking to add datasheet links to the BOM table, using KiCad V6:

How do I get this working with easyeda? also more than data sheet i want to create clickable url for references or values which takes to that specific product buy link.? any wrong around or suggestions?

qu1ck, 2023-01-11:
Urls are detected automatically now since 39fc0c466cf1bd98464efae56c9713aaddbeb5a3 as long as it's the only thing in your field. 

But extra fields are not supported in easyeda. I see that they now added support to put custom fields in each component but there is no convenient way to bulk edit all components because BOM table doesn't show custom fields.

If anyone wants to add support to parse custom fields in easyeda parser I'll review it but I don't have time to add it right now.

itsboo07, 2023-01-11:
> If anyone wants to add support to parse custom fields in easyeda parser I'll review it but I don't have time to add it right now.

thanks alot. please update me when you get it working.

qu1ck, 2023-01-11:
I filed #348 to keep track of it.

commit 04f4afbe (2019-06-02): Add datasheet field as extra field

Fixes #82