openscopeproject/InteractiveHtmlBom #84: Check boxes clear when switching views

ThomasHulme, 2019-03-19:
When switching between _BOM only_, _BOM left drawings right_ or _BOM top drawings bottom_, all the check boxes are cleared. This also happens when switching between _front view_, _bottom view_ or _both_.

Generated in KiCad 5.1
Shown in Chrome 73.0.3683.75

qu1ck, 2019-03-19:
I can't reproduce this on linux, same exact version of Chrome.

What version of the plugin are you using?
Do checkboxes reappear if you refresh after switching bom view? Does bom view reset to default after you refresh or does it persist? 
Are you viewing bom page from local file or some special location like nfs share?

ThomasHulme, 2019-03-19:
I'm on Windows 10.
I downloaded the plugin 3 days ago so I'm assuming its the latest version.
Check boxes are always visible its just the tick within them that clears. Refreshing the page sets everything back to default, including the dark theme setting.
Viewing the html page from C Drive.

qu1ck, 2019-03-19:
Does your board name have any special characters in it? Can you share the board itself? Feel free to remove all contents except board outline.

Also it would help if you took screenshot of Chrome app storage. To do that press `f12` in chrome, go to `Application` tab and click on `Local Storage` -> `file://`. Put `KiCad_HTML_BOM__<your board name> in filter and take screenshot of the contents. While you are at it, do you see the values there change when you tick checkboxes or change bom view settings?

ThomasHulme, 2019-03-19:
[image]
No special characters

[image]
There's no values at all.

Do you know if this is a setting in Chrome to stop local storage?

qu1ck, 2019-03-19:
That is weird. To my knowledge there is no such setting in Chrome.

What do you get if you enter `window.localStorage.getItem("blank")` in `Console` tab of chrome dev tools? Should be `null`.

Can you check with any other board (like KiCad demo) and in Firefox? Not Edge or IE, those don't enable local storage.

If this happens only with this particular board I will need the kicad_pcb file to debug further. Again, you can remove all footprints/traces/graphics from it if you can't share details of your design.

ThomasHulme, 2019-03-19:
Console returns:
Uncaught DOMException: Failed to read the 'localStorage' property from 'Window': Access is denied for this document.
    at <anonymous>:1:8

Just tried the same file in Firefox and all is working OK. Seems its just Chrome that's not working correctly.

Checking local storage on other website reveals their storage values, so I'm guessing this has to do with permissions.

qu1ck, 2019-03-19:
Check Chrome settings -> Content Settings -> Cookies
Do you have block setting on there? I think Chrome conflates cookies and local storage there.

ThomasHulme, 2019-03-19:
Yep, works now.
Set Chrome to allow all cookies (for now) and it works, didn't think of this before as I was looking for local storage settings.

Don't think i can add it to the exception list though due to the "URL" always being different for each file.

Thanks for helping.

qu1ck, 2019-03-19:
No worries. I actually found another config related bug while debugging this so thanks for reporting :)

closed 2019-03-19 by qu1ck