openscopeproject/InteractiveHtmlBom #87: Can't click on a component through a virtual component

dhaillant, 2019-03-26:
Hi,
I have a board which goes behind a front panel. I use a virtual component in Pcbnew to help me place the controls (potentiometers, etc).
This virtual component is bigger than the board itself and it "recovers" the other components.

Now, if that front panel component is set as "Through hole", when I click on the ibom map, I'm able to select a component (it's either the front panel that is selected or the component "behind" it).

But, if I set the front panel as a "Virtual" component, I can't select the components "behind" it. The check box "blacklist virtual components" is set but still, it avoids selecting components "behind" it.

I have to delete the virtual component in order to be able to click on the components in the map view.

Thank you
David

closed 2019-03-27 by qu1ck

qu1ck, 2019-03-27:
Thanks for bug report, let me know if fix works for you.

dhaillant, 2019-03-27:
It works! Thank you very much!

commit 1668fdc6 (2019-03-27): Fix canvas clicks stuck on modules w/o handlers

Fixes #87