openscopeproject/InteractiveHtmlBom #88: Add an option to select a component from the list by clicking instead of hovering

dhaillant, 2019-03-27:
I'd like to have an option that could let me select a component, from the list, only by clicking on it, not by simply hovering it.

With "hovering selection", if you slightly move your mouse over the list, the map highlights some other component. This can mislead you when assembling your board.

Thank you!
David

qu1ck, 2019-03-27:
I'll keep this low priority since there are already shortcuts for bom navigation. Just move mouse away from the table and use keyboard if you want to avoid accidentally switching active row.

EeliK, 2020-03-23:
> Just move mouse away from the table and use keyboard if you want to avoid accidentally switching active row.

That assumes quite much about the working environment and other restrictions of the user. I have difficulties with this issue.

qu1ck, 2020-06-27:
This is now possible using custom js.
Create `user.js` file in `web` directory with following contents:
```javascript
EventHandler.registerCallback(IBOM_EVENT_TYPES.BOM_BODY_CHANGE_EVENT, () => {
    for(var tr of bom.childNodes) {
        tr.onclick = tr.onmousemove;
        tr.onmousemove = null;
    };
});
```

Now highlight will be bound to click event instead of hover.

closed 2020-06-27 by qu1ck

mondalaci, 2023-08-16:
I have created `~/InteractiveHtmlBom/web/user.js` with the above content, generated a new BOM, and the selection is still hover-based.

What am I missing?

(I'm using the 2.7.0 InteractiveHtmlBom.zip release.)

qu1ck, 2023-08-16:
How are you running the plugin?

mondalaci, 2023-08-16:
I'm unsure what you're asking exactly, but I'm using it from KiCad 7.0.7 running on Linux Mint.

qu1ck, 2023-08-16:
So you are clicking the button (or launching from menu) in Pcbnew? If so then most likely you did not put hte user.js in correct path because `~/InteractiveHtmlBom` is not a path that kicad uses for plugins, unless you made a manual symlink.

You should put `user.js` where plugin is installed. If you installed it from kicad's plugin manager then you can get to that directory by clicking "Open package directory" in plugin manager. Navigate to plugin's folder there and then to `web` subfolder.

mondalaci, 2023-08-16:
That did the trick, thanks! The correct directory on my system is `~/.local/share/kicad/7.0/scripting/plugins`