openscopeproject/InteractiveHtmlBom #89: Chinese footprint name is not supported

zhqsoft, 2019-04-07:
Chinese footprint name is not supported

qu1ck, 2019-04-07:
I can not reproduce this issue. Add more details and repro steps, if possible share the board file that shows the issue.

zhqsoft, 2019-04-08:
If the footprint name value has Chinese characters, IBOM file cannot be exported .
I use the latest unpublished tools. 
Version V2.0 release is OK 

[image]
[image]

[CN_name_TEST.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/3052882/CN_name_TEST.zip)

qu1ck, 2019-04-08:
I see, yeah that is because of recently added fab layer support. Will fix soon.

commit 07b8da79 (2019-04-08): Add fallback for characters not in newstroke font

Fixes #89
closed 2019-04-08 by qu1ck