openscopeproject/InteractiveHtmlBom #91: Rendering at native resolution

JJJollyjim, 2019-04-09:
I noticed that the board was slightly blurry on my phone (where `window.devicePixelRatio = 2.608...`). I assumed this was because the code rendered at a 1:1 dom-pixel to canvas-pixel ratio, but it turns out the ratio is 2:1.

Is there a reason for this doubling, other than improving the rendering on `devicePixelRatio==2` devices? In other words, should I change the ratio to `devicePixelRatio`:1 or `2*devicePixelRatio`:1?

I imagine the former option would significantly improve performance on lower-resolution devices.

qu1ck, 2019-04-09:
I did this mainly to improve fidelity on high dpi screens (macbook). I never really optimized for phones with their non-integer scaling. See #10.
Is performance really an issue? Or is it just blurriness?

If you just change canvas ratio the scroll-zoom and panning will likely misbehave.

JJJollyjim, 2019-04-09:
I have made the change in a copy of the source code, and changed all the relevant factors of 2 to factors of `devicePixelRatio` -- everything works perfectly. On my fairly good desktop (with `devicePixelRatio=1` I get ~9fps dragging the motherboard demo around, so a performance bump could be nice. Unfortunately, this change ends up making no discernible performance difference anyway :)

I'll pull-request the change in a bit, if you're okay with it.

qu1ck, 2019-04-09:
I suspect this doesn't make any difference because browsers optimize for actual pixel size of canvas irrespective of virtual size. Even when canvas ratio is manipulated to be low no blockiness or pixelization appears, which means rendering engine still operates on physical pixels (at least).

But it's probably more technically correct to use the devicePixelRatio so I'll merge the patch assuming it works on firefox and edge.

>  I get ~9fps dragging the motherboard demo around, so a performance bump could be nice. 

"Continuous draw on drag" toggle is there for exactly that purpose. There are thousands of draw calls to render that board, I'm surprised it's as fast as it is.

PR #92 (2019-04-09): Render 1:1 with device pixels
Resolves #91
closed 2019-04-09 by qu1ck