openscopeproject/InteractiveHtmlBom #95: EasyEDA Support

martincolladodotcom, 2019-04-17:
Thanks for this amazin project!
I was wondering if would be possible to use this plugin with the gerber files and bom generated by EasyEDA?

qu1ck, 2019-04-17:
Gerber and bom files don't have enough information.
Fortunately EasyEDA internal format is open and more or less documented. It should be possible to write an importer for that format. I'll have to investigate the details and see how to map necessary features to internal ibom format.

qu1ck, 2019-05-24:
Ok, this is implemented and works fairly well. There are some limitations:
- Extra fields are not supported yet.
- Component value, reference detection can be messed up. I have seen that with imported boards, native EasyEDA projects seem to work fine. If that happens with native board I would like to see your project.

Instructions are updated, simply run
```shell
python generate_interactive_bom.py your_board.json
```

[Demo](https://openscopeproject.org/InteractiveHtmlBomDemo/html/2.4G_Telecontrol_Board%20V1.0-PCB.html)

martincolladodotcom, 2019-05-24:
I'll test it this weekend with some projects. Thanks for this amazing feature!

closed 2019-07-05 by qu1ck

qu1ck, 2022-10-29:
You need to use fake X server like xvfb or define `INTERACTIVE_HTML_BOM_NO_DISPLAY` environment variable.

See https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Tips-and-Tricks for more info.