openscopeproject/InteractiveHtmlBom #97: When zooming in the preview,The Front preview shows slowly.

zhqsoft, 2019-05-06:
Is there a problem with version V2.1? When zooming in the preview ,The Front preview shows slowly.The bottom preview is good. 
I uploaded BOM files and PCB files 
[7381.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/3147283/7381.zip)

qu1ck, 2019-05-06:
I opened your file and don't notice any issues. Do you mean the rendering is slow? There are more elements on top preview than on the bottom but it's not particularly complex board. Can you open motherboard demo ok? 
Do you notice slowness only when zooming or when draggin as well?

What browser do you use?

zhqsoft, 2019-05-06:
I use EDGE browser, bad browser .Google Chrome browser can be used properly .

zhqsoft, 2019-05-07:
Another problem ,
The more electronic components, the thicker the silk screen rendering. 
[image]

qu1ck, 2019-05-07:
Thanks. Interestingly the issue was there for a long time, it just became slightly more noticeable with recent changes.

closed 2019-05-07 by qu1ck