openscopeproject/InteractiveHtmlBom #99: polygons on copper are shown offset

poeschlr, 2019-05-15:
eagle import can generate pure polygons on the copper layer. These are displayed at the wrong position. I used ibom v2.1 with kicad 5.1.0 under windows 10.

Sadly i can not share the full design. I can however share the footprint to play with:

(module PG-USON-10 (layer F.Cu) (tedit 0)
  (fp_text reference U$1 (at -2 2 90) (layer F.SilkS) hide
    (effects (font (size 0.9652 0.9652) (thickness 0.08128)) (justify left bottom))
  )
  (fp_text value M3702USON10 (at 3 2 90) (layer F.Fab) hide
    (effects (font (size 0.9652 0.9652) (thickness 0.08128)) (justify left bottom))
  )
  (fp_poly (pts (xy -1.2 0.8) (xy 1.2 0.8) (xy 1.2 -0.8) (xy -1.2 -0.8)) (layer F.Cu) (width 0))
  (fp_poly (pts (xy -1.3 0.9) (xy 1.3 0.9) (xy 1.3 -0.9) (xy -1.3 -0.9)) (layer F.Mask) (width 0))
  (fp_poly (pts (xy -1.1 -0.1) (xy -0.1 -0.1) (xy -0.1 -0.7) (xy -1.1 -0.7)) (layer F.Paste) (width 0))
  (fp_poly (pts (xy 0.1 -0.1) (xy 1.1 -0.1) (xy 1.1 -0.7) (xy 0.1 -0.7)) (layer F.Paste) (width 0))
  (fp_poly (pts (xy 0.1 0.7) (xy 1.1 0.7) (xy 1.1 0.1) (xy 0.1 0.1)) (layer F.Paste) (width 0))
  (fp_poly (pts (xy -1.1 0.7) (xy -0.1 0.7) (xy -0.1 0.1) (xy -1.1 0.1)) (layer F.Paste) (width 0))
  (fp_line (start -1.3 2) (end 1.5 2) (layer F.SilkS) (width 0.1))
  (fp_line (start 1.5 2) (end 1.5 -2) (layer F.SilkS) (width 0.1))
  (fp_line (start 1.5 -2) (end -1.5 -2) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.5 -2) (end -1.5 1.8) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.5 1.8) (end -1.5 1.9) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.5 1.9) (end -1.5 2) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.5 2) (end -1.3 2) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.3 2) (end -1.3 1.9) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.3 1.9) (end -1.3 1.8) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.3 1.8) (end -1.5 1.8) (layer F.SilkS) (width 0.1))
  (fp_line (start -1.5 1.9) (end -1.3 1.9) (layer F.SilkS) (width 0.1))
  (fp_text user . (at -1.5 2.5) (layer F.SilkS)
    (effects (font (size 1.2065 1.2065) (thickness 0.1016)) (justify left bottom))
  )
  (pad P$1 smd rect (at -1 1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$2 smd rect (at -0.5 1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$3 smd rect (at 0 1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$4 smd rect (at 0.5 1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$5 smd rect (at 1 1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$6 smd rect (at 1 -1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$7 smd rect (at 0.5 -1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$8 smd rect (at 0 -1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$9 smd rect (at -0.5 -1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
  (pad P$10 smd rect (at -1 -1.4) (size 0.25 0.8) (layers F.Cu F.Paste F.Mask)
    (solder_mask_margin 0.0762))
)

poeschlr, 2019-05-15:
Ok i now updated to 5.1.2 and it works. Sorry for the noise

closed 2019-05-15 by poeschlr

poeschlr, 2019-06-27:
Updating was not the fix. Closing pcb_new and reopening after eagle import is what is required.

qu1ck, 2019-06-27:
Thanks for the info, that's a curious pcbnew behavior. 
If you can provide a simple repro eagle file so that I can try and import it and see what changes in pcbnew after it's reopened I may be able to fix this upstream or at least report a bug with more info.