openscopeproject/InteractiveHtmlBom #100: Some components not coming through…?

hedefalk, 2019-05-22:
Sorry to post as bug, but I thought this be better asked here than the forum. I'm having issues with four pots not coming through. Everything else comes out fine, but these pots do not. They do come out with builtin bom export and I can't seem to understand what's special about them:

[image]

These are missing:

1 | OFFSET2,OFFSET1,CV_ATT2,CV_ATT1 | Alps_RK09K_Hor_centered | 4 | B50K
-- | -- | -- | -- | --

[image]

The only thing half-special about them I can find is that they where auto-generated from a single hierarchical sheet used twice from main sheet, but so are on the other hand these too:

12 | SYM_ADJ2,SYM_ADJ1 | Potentiometer_Triwood_RM-065 | 2 | B50K
-- | -- | -- | -- | --

I'm no super sure how this works but the schematic symbol in the reused sch file looks like this:

```
$Comp
L Device:R_POT CV_ATT1
U 1 1 5CBD5721
P 2050 2400
AR Path="/5CBA7864/5CBD5721" Ref="CV_ATT1"  Part="1" 
AR Path="/5CC101A1/5CBD5721" Ref="CV_ATT?"  Part="1" 
AR Path="/5CC1B59D/5CBD5721" Ref="CV_ATT2"  Part="1" 
F 0 "CV_ATT1" H 1981 2354 50  0000 R CNN
F 1 "B50K" H 1981 2445 50  0000 R CNN
F 2 "viktors-footprints:Alps_RK09K_Hor_centered" H 2050 2400 50  0001 C CNN
F 3 "~" H 2050 2400 50  0001 C CNN
	1    2050 2400
	1    0    0    1   
$EndComp
```

I don't think I myself gave any question marks to them from the start, but that binding is added automagically to _just work_ for hierarchical sheets. To me it looks very similar to the working case (a trimpot): 

```
$Comp
L Device:R_POT_TRIM SYM_ADJ1
U 1 1 5CBD56D9
P 6800 5550
AR Path="/5CBA7864/5CBD56D9" Ref="SYM_ADJ1"  Part="1" 
AR Path="/5CC101A1/5CBD56D9" Ref="SYM_ADJ?"  Part="1" 
AR Path="/5CC1B59D/5CBD56D9" Ref="SYM_ADJ2"  Part="1" 
F 0 "SYM_ADJ1" H 6730 5504 50  0000 R CNN
F 1 "B50K" H 6730 5595 50  0000 R CNN
F 2 "viktors-footprints:Potentiometer_Triwood_RM-065" H 6800 5550 50  0001 C CNN
F 3 "~" H 6800 5550 50  0001 C CNN
	1    6800 5550
	-1   0    0    1   
$EndComp
```

Maybe I'm missing something here…

Happy to post files!

qu1ck, 2019-05-22:
Most likely pots that don't show up are marked as virtual components. Either make them normal in pcbnew or untick "blacklist virtual" in ibom settings.

If that is not the case please share the pcb file.

hedefalk, 2019-05-22:
@qu1ck No, sorry, they're not virtual. kicad_pcb file enough?

hedefalk, 2019-05-22:
https://gist.github.com/hedefalk/09f4da739c6f646273b33414e7bf124b

qu1ck, 2019-05-22:
Your pots are on the front side of the pcb :) Switch to 'FB' or 'F' and they will be in the table.

hedefalk, 2019-05-22:
@qu1ck doh! Man do I feel stupid. Thx!

closed 2019-05-22 by hedefalk