openscopeproject/InteractiveHtmlBom #101: Is it possible to build the bom from just gerbers and position files?

mkschreder, 2019-05-27:
Is it possible to build the interactive view from just gerbers and component position files allowing one to search for components?

qu1ck, 2019-05-27:
No, gerber files have no information about components, only about pcb layers.

mkschreder, 2019-05-27:
The placement file has information about components.

mkschreder, 2019-05-27:
It should be possible to at least reconstruct the placement of references.

mkschreder, 2019-05-27:
I started looking at it and what I have done so far is export gerbers from
gerbview to kicad pcb format and for now without the placement data. The
interactive bom script says it failed to parse the file. Possibly because
there is no module data in the file. I have no time to look further into it
but I'm thinking I could use kicad-library-utils and parse the placement
data and then create dummy modules in the kicad pcb file and try exporting
again. Would be very helpful because I have old gerbers with placement
files that I want to browse this way.

qu1ck, 2019-05-27:
If you provide examples of files I can look into it more but I highly doubt placement files have anything other than positions of component centers and some part id (not board component reference) which is not enough. At the minimum bom needs board reference, value and footprint.

> The interactive bom script says it failed to parse the file. 

How are you running it and what is the exact error message?

qu1ck, 2019-06-01:
I will close this for inactivity unless I hear from you in a few days.

closed 2019-06-06 by qu1ck