openscopeproject/InteractiveHtmlBom #103: Comma in field name or field value

rusefi, 2019-06-27:
Unfortunately we use commas in extra field names and values
```
      (fields
        (field (name MFG,MFG#) Matsushita,ERJ6GEY0R00V)
        (field (name VEND1,VEND1#) DIGI,P0.0ADKR-ND))
``` 

and I am failing to find a way around this.

By default I use 
```
%ibom_cmd% --extra-fields "MFG,MFG#","VEND1,VEND1#" frankenso.kicad_pcb --netlist-file frankenso.net
```
and this produces separate MFG and MFG# columns without values.

I've tried hacking it with sed to make it
```
        (field (name MFG_MFG#) Matsushita,ERJ6GEY0R00V)
        (field (name VEND1_VEND1#) DIGI,P0.0ADKR-ND))
```
but this case I get an error message about error parsing net file :(

Attached are complete files if you would have time to try to reproduce this
[issue.zip](https://github.com/openscopeproject/InteractiveHtmlBom/files/3332777/issue.zip)

qu1ck, 2019-06-27:
Good news is this works fine from GUI. Unless you need automation that is good workaround.
I'll come up with some kind of escaping for commas for cli.

closed 2019-06-27 by qu1ck

qu1ck, 2019-06-27:
See example of escaping in commit message or wiki.

rusefi, 2019-06-27:
Thank you! Yes, there is automation and it works now :)

commit 872cdf4d (2019-06-27): CLI parsing for fields with comma delimiter supports escaping

Example:
--extra-fields "MFG\,MFG#,VEND1\,VEND1#"
will yield fields with names
MFG,MFG#
VEND1,VEND1#

Fixes #103