openscopeproject/InteractiveHtmlBom #104: Bounding box of rotated parts

shufps, 2019-07-22:
Wow, I love this plugin sooo much 😍 Great work!

I figured out, it would be nice to also rotate the bounding box of a part if the part itself is rotated.

[image]

Also it would be nice if the selection in the part list would be clickable - so it doesn't deselect the selected part if the mouse is moved.

Just my two observations with fresh eyes :)

Keep up the awesome work! 😍

qu1ck, 2019-07-22:
Bounding boxes are calculated that way in pcbnew, it's not currently done by plugin. I suppose I could calculate them manually and rotate appropriately but that will make click detection quite a bit harder.

Regarding clicking on rows, there is #88 already.

closed 2019-10-21 by qu1ck

shufps, 2019-10-22:
wow! thank you thank you thank you <3