openscopeproject/InteractiveHtmlBom #108: Starting with two monitors, using iBom on 2nd one but not running for the moment: hibernate laptop, restart with single monitor (1st) - you can't get iBom to show any more on 1st monitor

kenneth558, 2019-08-20:
Linux Mint/Firefox.  Starting with two monitors, using iBom on 2nd one but not running for the moment: hibernate laptop, restart from hibernation with only 1st monitor - you can't get iBom to show any more on 1st monitor.  I would think that moving pcbnew to 1st monitor would have brought iBom over also, but didn't happen.  Without seeing iBom window, I don't know how to close it - it doesn't show as a separate task from pcbnew with the basic right-click gui commands to do window closes and moves.  Then with iBom running, you can't even close Pcbnew, nor KiCad at all.  I suppose I could use CLI to close things, but I'm just bringing this situation to your attention:  KiCad is totally not able to be closed with the normal gui means, except with System Monitor.  My gui is cinnamin.

qu1ck, 2019-08-20:
If I understand correctly you are saying that when you launch the plugin in 2 monitor system, leave it open, hybernate, disconnect second monitor and restart with only 1 monitor, if ibom dialog was on the now non-existant monitor it becomes unresponsive, invisible and blocks interactions with kicad UI.

That is somewhat expected behavior for all dialog windows. Unless you setup your window manager to automatically move all windows to existing monitor when monitor configuration changes. I believe cinnamon can do that, if not there will be a third party app for that.

Here is a simple workaround: focus pcbnew window (which will in turn focus ibom dialog) and press `enter`. That will invoke `Generate BOM` action so dialog will close. 

Also I just committed a fix so that standard `Escape` key will invoke cancel button so you can use that too.

kenneth558, 2019-08-21:
> That will invoke Generate BOM action so dialog will close.

I've power cycled the laptop since then and am just making a quick reply here, but I wonder if the above workaround would bring the iBom window back to the laptop monitor on next invoking.  I suspect it would not because all that happens there is iBom open-close.  I think it would need a monitor change somehow between open and close would need to happen for iBom to open in the live monitor.  Is there a simple way on IBom dialog open to ensure it is open on a live or existing monitor?  (Even though that really ought to be an OS responsibility)

qu1ck, 2019-08-21:
Ibom window is setup as a dialog window, which means it will always open on
the active monitor. If pcbnew is on active monitor (and it should be since
otherwise you will not be able to start the plugin) the dialog will open
centered on the same monitor.

If for some reason that doesn't happen that would be a bug in wxwidgets or
in window manager incorrectly detecting/updating monitor configuration.
Either way that is not something I can fix on my end.

You can test this easily when you have 2 monitors connected. Call plugin in
one monitor and observe the dialog window opening on same one where pcbnew
is. Then close it and move pcbnew to second monitor, call plugin again.
Dialog should appear on second (active) monitor.

kenneth558, 2019-08-21:
> Ibom window is setup as a dialog window
That settles it.

BTW, I assume that is what python calls a dialog window?  Not c++ or something else, right?  I thought maybe I'd delve into it a little on my own....

qu1ck, 2019-08-21:
Plugin uses wxPython GUI library, which is just a python wrapper around wxWidgets which (on most platforms and most UI elements) is a wrapper around native GUI controls. So it's not technically correct to say that it's python dialog or c++ dialog window, in the end it's a system native window.
But yes, topmost call to create the dialog is in python plugin code.

To add a bit more detail: KiCad uses wxWidgets which allows python/wxPython plugins to live in the context of the same application. So when you launch plugin from pcbnew it lives in the parent process' context and uses same UI styling etc. From system's perspective it's a single cpp app, even though parts of it are running py code on cpython interpreter.

closed 2019-08-22 by qu1ck