openscopeproject/InteractiveHtmlBom #109: Python exception

Topher-Holmes, 2019-08-30:
Running KiCAD 5.1.4 on Windows 10 Home I get a Python exception dialog every time I run the script.  I've tried a few different projects with the same results.  
I run the script from Pcbnew Tools -> External Plugins.

[image]

I tried getting a stack trace, but it doesn't work.  Sorry, complete newbie here.  

Py 0.9.8
Python 2.7.15 (default, Oct 11 2018, 12:09:51)  [GCC 8.2.0 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: C:\Users\chris\AppData\Roaming\kicad\PyShell_pcbnew_startup.py
from InteractiveHtmlBom import plugin
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ImportError: cannot import name plugin

qu1ck, 2019-08-30:
Looks like plugin failed to parse your netlist or xml file. Try regenerating it, if the issue persists I would need to look at your project files to determine what fails.
But either way there should be proper error handling and message for this case and not a crash. I'll fix that.

Topher-Holmes, 2019-08-31:
Regenerating the netlist fixed it.  The BOM generates OK now.  Thanks for the quick reply!

closed 2019-08-31 by Topher-Holmes